bitfocus/companion-module-raspberry-gpio #1: GPIO pins on Raspberry Pi

KrisMcC1, 2020-11-02:
Hi lovely people

It would be great if a module existed to access the GPIO pins on a raspberry pi that companion is running on. As an extra bonus it would be extra amazing if you could access GPIO of other Pis on the network.

What do we think?

K

melimat, 2020-11-02:
HI all, so currently the module is capable of interfacing with CPIO pins on RPi. It delivers also state variables for each pin, state feedbacks and presets. Sadly though I did not implement control over network. So the module has to run physicaly in CompanionPi. But the network control is doable, so it's on my TODO list. But probably there should be a separate module for that.

tom-niddatal, 2022-11-28:
Hi guys,

I tried to install GPIO on my Raspi. It was a little bit a hustle, but at least, I see that connection now in my connections section. Trying to add this connection fails with the following message:

Failed to create connection
Failed: TimeoutError: Timed out after 5s.

Have you got any idea, what I am doing wrong?

guiguinux, 2023-01-17:
> Hi guys,
>
> I tried to install GPIO on my Raspi. It was a little bit a hustle, but at least, I see that connection now in my connections section. Trying to add this connection fails with the following [...]

+1

XENONChromatic, 2024-01-30:
Bumping this.

istnv, 2024-03-23:
GPIO module updated for v3

closed 2024-03-23 by istnv