bitfocus/companion-module-raspberry-gpio #5: Maximum pin limit?

registerednerd, 2021-05-08:
Everything seems to work fine when I have up to about 4 pins enabled in the module, but once I try to add the 6 pins that I need for the project at hand, it stops adding new variables to the instance. If I reboot the Pi after ending up in this bind, then all of the variables disappear, almost as if the instance isn't configured at all (but my settings are still in place when I view the instance settings). 

Running a fresh install on a Pi4 8GB, for reference.

istnv, 2021-05-08:
Which pins are you trying to use?

registerednerd, 2021-05-08:
13, 16, 19, 20, 21, 26. If I just add 13, 16, 19, 20, it seems to work fine. Haven't tried 21 and 26 in isolation (trying now)

istnv, 2021-05-08:
The module does not prevent using pins that the Pi uses for internal purposes.

registerednerd, 2021-05-08:
Looks like it's something specific to pin 26. AFAIK, though, I don't have anything else configured using those pins. That at least gives me a direction to go with troubleshooting. I picked those particular pins so that I could get the connections I needed in a single custom DuPont connector, but I do see that there's another block that actually fits the need better, I just need to re-lay-out my connector.

closed 2024-03-23 by istnv