bitfocus/companion-module-raspberry-gpio #10: Failed: TimeoutError: Timed out after 5s.

ckmiecik, 2023-05-18:
I have finally gotten the connection to show up (there is no documentation on how to install this so that would be helpful in the readme file maybe?). I followed the directions below from another post, but now when I try to add the connection I get the following message 

Failed: TimeoutError: Timed out after 5s.

I see there are other people commenting that they are getting the same message so I thought I would start a new issue to see if we can get this figured out.  Here are the steps I followed from another person's post

1 - new SD card created from the 2.4.1 companion pi image
2 - $ cd /usr/local/src/companion
3 - $ sudo git submodule add -f https://github.com/bitfocus/companion-module-raspberry-gpio.git module-local-dev/raspberry-gpio
4 - $ cd module-local-dev/raspberry-gpio
5 - $ -- Manually install nvm v14.21.2 (NOT THE LATEST VERSION)
6 - $ sudo apt install npm
7 - $ npm install -g yarn
8 - $ yarn add onoff
9 - $ sudo reboot

jmcontra, 2023-10-27:
I don't know if you're still interested, but I found the issue.  There's a issue with permissions on the /usr/local/src/companion/module-local-dev/raspberry-gpio folder which prevents the dependencies from being installed.

to fix the permissions you can run 

$ sudo chown -R $USER /usr/local/src/companion/module-local-dev/raspberry-gpio

followed by 

$ npm install

Which will install the dependancies called for by the package.json file and everything should work. 

Also, you said "Manually install nvm v14.21.2"  which threw me off for a while. it's node v14.21.2 not nvm v14.21.2.  for anyone following along at home, run

$ curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.33.6/install.sh | bash
$ export NVM_DIR="$HOME/.nvm"
[ -s "$NVM_DIR/nvm.sh" ] && \. "$NVM_DIR/nvm.sh"  # This loads nvm
[ -s "$NVM_DIR/bash_completion" ] && \. "$NVM_DIR/bash_completion"  # This loads nvm bash_completion
$ nvm install v14.21.2

and then continue with installing npm.

Julusian, 2024-04-20:
This is fixed now that the module is included in the beta builds

closed 2024-04-20 by Julusian