bitfocus/companion-module-raspberry-gpio #13: GPIO crashes on Rpi4

RobJLocke, 2024-03-16:
GPIO module installed in /opt...
web interface warns of problems - module crashes
Log reads
error: node:internal/errors:496
    ErrorCaptureStackTrace(err);
    ^

Error [ERR_MODULE_NOT_FOUND]: Cannot find package '@companion-module/base' imported from /opt/companion/module-legacy/manifests/companion-module-raspberry-gpio/lib/main.js
    at new NodeError (node:internal/errors:405:5)
    at packageResolve (node:internal/modules/esm/resolve:890:9)
    at moduleResolve (node:internal/modules/esm/resolve:939:20)
    at defaultResolve (node:internal/modules/esm/resolve:1132:11)
    at nextResolve (node:internal/modules/esm/loader:163:28)
    at ESMLoader.resolve (node:internal/modules/esm/loader:835:30)
    at ESMLoader.getModuleJob (node:internal/modules/esm/loader:424:18)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/module_job:77:40)
    at link (node:internal/modules/esm/module_job:76:36) {
  code: 'ERR_MODULE_NOT_FOUND'
}

Node.js v18.18.1

Not sure what to do here

istnv, 2024-03-17:
The base module has not been packaged so the libraries needed are not included.
You will need to install `yarn`:
 `sudo npm install -g yarn`
Then in the `raspberry-gpio` directory:
`sudo yarn`

The required libraries will be downloaded and installed.
Your Pi needs to be online for this.

RobJLocke, 2024-03-17:
npm and yarn installed. When I run yarn in the raspberry-gpio directory

 

Syntax Error: “Unexpected end of JSON input”  in ~manifests/package.json. This is a blank document/file.

 

Tried one or two other directories – same result.

 

Companion log reads:

 

error: node:internal/errors:496

    ErrorCaptureStackTrace(err);

    ^

 

Error [ERR_MODULE_NOT_FOUND]: Cannot find package ***@***.***/base' imported from /opt/companion/module-legacy/manifests/companion-module-raspberry-gpio/lib/main.js

    at new NodeError (node:internal/errors:405:5)

    at packageResolve (node:internal/modules/esm/resolve:890:9)

    at moduleResolve (node:internal/modules/esm/resolve:939:20)

    at defaultResolve (node:internal/modules/esm/resolve:1132:11)

    at nextResolve (node:internal/modules/esm/loader:163:28)

    at ESMLoader.resolve (node:internal/modules/esm/loader:835:30)

    at ESMLoader.getModuleJob (node:internal/modules/esm/loader:424:18)

    at ModuleWrap.<anonymous> (node:internal/modules/esm/module_job:77:40)

    at link (node:internal/modules/esm/module_job:76:36) {

  code: 'ERR_MODULE_NOT_FOUND'

}

 

Node.js v18.18.1

 

Something not right somewhere, but novice level in linux or js. Have I missed something earlier in the process?

R

istnv, 2024-03-17:
I missed what directory where you had installed the module, sorry. 
Remove it from the `/opt/companion/module-legacy` directory, that will cause the errors you are getting.
You want to clone the module into `/opt/companion-module-dev/raspberry-gpio`:

```
$ cd /opt/companion-module-dev
$ sudo git clone https://github.com/bitfocus/companion-module-raspberry-gpio.git raspberry-gpio

```

Then `cd` into the new `raspberry-gpio` directory and run the `yarn` sequence I posted earlier.

RobJLocke, 2024-03-17:
That seemed to work – Pi GPIO appeared in list of “good” modules. Starts and has config page

 

The final issue now is one of:

system: ** Connection started **

Starting up module class: GpioInstance 

Sentry disabled 

Module-host accepted registration 

error: GPIO is inaccessible

 

pigpio is installed. Tried the remote GPIO in raspi-config on and off.

Companion is already a member of gpio group.

 

Last hurdle I think , but not sure what else to try

 

R

istnv, 2024-03-17:
The only way I cleared the no access error was to rebuild `onoff`
`cd` to the repository directory, then:
`$ sudo yarn add onoff`

RobJLocke, 2024-03-18:
Could you clarify repository directory. Is it within the  / opt structure or elsewhere?  Sorry not very familiar with where programme parts end up in linuxSent from Outlook for Android

RobJLocke, 2024-03-18:
A typo on my part. Many thanks for all the help. I have a green tick, log has info only lines. Now to the physical build....BestSent from Outlook for Android

closed 2024-03-23 by istnv