bitfocus/companion-module-raspberry-gpio #14: Linking Raspi GPIO state to button

RobJLocke, 2024-03-20:
Pretty new to Companion but so far have managed to run CompanionPi headless on Pi4 4GB, install the GPIO module. I am hoping to link a simple contact closure to a Blackmagic videohub for camera preview. I have set up 4 momentary buttons from the BMD presets (work on test). Feedbacks (for specific pins) show contact closures (button colour change), but no actions are being forwarded to the BMD device itself. I'm guessing it is some kind of internal link but at I am at a loss to find the right method.

istnv, 2024-03-25:
Feedback on a button is only able to change the appearance of the button (text/colors/graphics).
Add a trigger that watches that feedback event and add your desired BMD actions there.

RobJLocke, 2024-03-25:
Got it. Companion needs a restart after setting new values in triggers. I was going around in circles until I realised this- nothing seemed to work.  Is this normal on CompanionPi?.Sent from Outlook for Android

istnv, 2024-03-25:
Restart should not be necessary, but if it works :/

closed 2024-07-27 by istnv