bitfocus/companion-module-raspberry-gpio #15: Raspberry Pi IO control

mark-epstein, 2024-03-20:
Please consider a module to allow Companion to talk directly to the IO of the host computer, 

My vision of what this module would do is pretty basic: It would simply include commands to turn a specific output pin on or off; or read the state of a specified input pin.

Use cases:
My immediate goal is to use low voltage to control an AC circuit, so that Companion can reliably turn my equipment on and off unattended. I know there are third-party devices that can manage such circuits but they add a lot of complexity/points of failure to the system, not to mention that electrical code regulations suggest I cannot just split a hard-wired circuit and plug it into a controllable outlet. (It's perfectly legal, on the other hand, to insert a certified relay.)

Also a low-voltage control such as is already built-in to the Pi would let me connect to all manner of previously-built electronic circuits and legacy devices that are otherwise unfriendly to remote control. For example, I have an amplifier that has a set of low-voltage contacts to temporarily mute its output. I'd like Companion to be able to control that, as the amplifier has no other control options.

I can (probably, haven't yet tried to) accomplish this with some minimal coding on the RPi and the generic http module. But the generic modules even admonish me to ask for a real module when the device is one that more users than myself are using. I think this fits.

istnv, 2024-03-20:
@Julusian Invite me to this repo. I have an update for this ready but need to finish setting up my pi dev system set up to work on a couple of issues that need deeper debugging.

Julusian, 2024-03-20:
done.

The work I haven't finished was rewriting/reworking the underlying `epoll` library to node-api and making sure we don't need to compile it. So there shouldn't be any conflicts

istnv, 2024-03-20:
gotcha.
I have a fix for a hard lockup if you enable a pin that is being used by the system and restart the module: ENOACCESS -> retry ->ENOACCESS -> retry ad nauseum.
Config tab never completes to disable the pin.
I may scan first to offer only the pins which could be accessed.

I'm having trouble on the config tab with some `isVisible` functions that always seem to return false.

The module works fine after aborting the restart cycle for a bad pin.

istnv, 2024-03-23:
@Julusian I was able to run this on a non-pi Ubuntu system.
Properly shows GPIO is not accessible error.
Won't work on a non-pi, but does not crash, either

closed 2024-03-23 by istnv