bitfocus/companion-module-raspberry-gpio #19: Error after 'companion-update' to latest beta

istnv, 2024-05-15:
After updating companion, the gpio module crashes:

```
system: ** Starting Connection from "/opt/companion/bundled-modules/raspberry-gpio/main.js" **
system: ** Connection started **
error: node:internal/modules/cjs/loader:1460
  return process.dlopen(module, path.toNamespacedPath(filename));
                 ^

Error: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /opt/companion/bundled-modules/raspberry-gpio/prebuilds/julusian-epoll-linux-arm64/node-napi-v8.node)
    at Module._extensions..node (node:internal/modules/cjs/loader:1460:18)
    at Module.load (node:internal/modules/cjs/loader:1203:32)
    at Module._load (node:internal/modules/cjs/loader:1019:12)
    at Module.require (node:internal/modules/cjs/loader:1231:19)
    at require (node:internal/modules/helpers:177:18)
    at I (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:415659)
    at /opt/companion/bundled-modules/raspberry-gpio/main.js:1:399013
    at 7107 (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:399133)
    at n (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:489818)
    at 8419 (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:405297) {
  code: 'ERR_DLOPEN_FAILED'
}

Node.js v18.20.2
system: ** Connection stopped **

Julusian, 2024-05-15:
What version of debian is it? `cat /etc/debian_version` will give the version number

istnv, 2024-05-15:
Debian 11.9
It was a companion pi image a couple of years ago that has been periodically updated.

Julusian, 2024-05-15:
hopefully v1.2.1 of this module will fix it, once it bubbles through to the builds

istnv, 2024-05-16:
Works fine, now. Thanks.

closed 2024-05-16 by istnv