bitfocus/companion-module-raspberry-gpio #20: RPi GPIO - Connection failure - Companion RPi v3.3.0.6991

carmino2k, 2024-06-04:
Hi everyone,
I recently installed the last Companion stable v3.3.0.6991 image on my raspberry Pi.
I tried installing the RPi_GPIO module, ok (in "status" I have the green check mark).
However, when I try to enable a Pin (ex: Gpio 5) in “Edit connection” I get an error:

“Connection Failure. Pin 5 is inaccessible Error: EINVAL: invalid argument, write”

I tried with other pins but same error.
I tried updating the “onoff” library to version 6.0.3 but I have the same error.

Can anyone help me ?
Thanks !!!

[image]

carmino2k, 2024-06-09:
same problem with version 3.3.1+223 (7000)

dani1810, 2024-06-19:
Same problem here.

I already checked the old issues and installed yarn and "onoff" manually. 
I also loaded the old version 3.2.2 of companion (without teh modul included) and loaded the module manually, but without any outcome. In the branch "fix/config" isn't such an error (only after reboot :( ), but before the reboot also not the variable for the GPIOs available.
Can anyone help and has any ideas? Is there actually a functionable combination of companion, the module and yarn (nmp/node)?
Thanks!

Brian-TV, 2024-07-21:
I ran across this issue with a RPi 5.  The GPIO's are no longer 0 to 27.  Mine were 573 to 598.  I went in and edited this file: /opt/companion/bundled-modules/raspberry-gpio/main.js    I then changed ";for(let t=0;t<=27;t++){const n=`use_gpio_${t}`;" to ";for(let t=573;t<=598;t++){const n=`use_gpio_${t}`;" and later on in the file I changed this line ";for(let e=0;e<=27;e++)" to ";for(let e=573;e<=598;e++)"

To find the GPIO numbers use this command: "cat /sys/kernel/debug/gpio |grep gpio"  

Hopefully they get this fixed, because any updates are going to wipe out the changed config.

Hope it helps the ones that need GPIO working now!!  Brian

istnv, 2024-07-21:
Is this with companion pi or did you install companion directly?

Brian-TV, 2024-07-21:
I installed the Companion Pi image. The latest stable one 3.3.1+7001

Brian-TV, 2024-07-21:
I installed the latest stable image 
[image]
[image]
version of Companion Pi 3.3.1+7001

istnv, 2024-07-21:
This is caused by a recent kernel update that offset the pins by 512.
Can you post the output of `raspi-gpio get`?

Brian-TV, 2024-07-21:
Raspi-gpio no longer works. It is now 'pinctrl'.

I can't SSH into that remotely from home. I'll have to post that later on. Mine seems to have moved up more than 512 here. 

If I'm remembering correctly, pinctrl addresses it as GPIO 0.

istnv, 2024-07-21:
I can add a check to the module but I need some non-root way to determine the offset.

Julusian, 2024-07-21:
Trying some things on my satellite pi (a zero w2)
```
julus@satellitepi:~ $ sudo cat /sys/kernel/debug/gpio 
gpiochip0: GPIOs 512-565, parent: platform/3f200000.gpio, pinctrl-bcm2835:
```
so it is definitely affected by this, but pinctrl doesn't seem to indicate it:
```
julus@satellitepi:~ $ pinctrl
 0: ip    -- | hi // ID_SDA/GPIO0 = input
 1: ip    -- | hi // ID_SCL/GPIO1 = input
```

There is a file `/sys/class/gpio/gpiochip512/base` which contains simply `512`. that file is readable by all
This is the only `gpiochip*` named folder inside of `/sys/class/gpio`

Compared to my home assistant pi:
```
julus@hass:/sys/class/gpio $ sudo cat /sys/kernel/debug/gpio 
gpiochip0: GPIOs 0-57, parent: platform/fe200000.gpio, pinctrl-bcm2711:
```
and 
```
julus@hass:~ $ cat /sys/class/gpio/gpiochip0/base 
0
```
There is /sys/class/gpio/gpiochip0 and /sys/class/gpio/gpiochip504 on this pi, which matches the two entries reported by `sudo cat /sys/kernel/debug/gpio` 

So I think the presence of `/sys/class/gpio/gpiochip512/base` is probably what this should be looking for.

istnv, 2024-07-21:
@Julusian 
I just found those files (which are readable).
On my Pi4 I'm seeing 2 chips in `/sys/class/gpio`: `gpiochip0` and `gpiochip504`
Running this:

```
find /sys/class/gpio -name 'gpiochip*' | while read f; do
 l=`cat ${f}/label`
 b=`cat ${f}/base`; 
 echo ${f} ":" ${l} "@" ${b}; 
done
```
returns:
```
/sys/class/gpio/gpiochip504 : raspberrypi-exp-gpio @ 504
/sys/class/gpio/gpiochip0 : pinctrl-bcm2711 @ 0

```
As long as the label has `pinctrl`, the number in `/base` could be retrieved.
The latest report above from a Pi5 starts at 573.

Brian-TV, 2024-07-22:
I ran your find script and here are the results from my Pi 5..

```
/sys/class/gpio/gpiochip512 : gpio-brcmstb@107d508500 @ 512
/sys/class/gpio/gpiochip565 : gpio-brcmstb@107d517c20 @ 565
/sys/class/gpio/gpiochip571 : pinctrl-rp1 @ 571
/sys/class/gpio/gpiochip548 : gpio-brcmstb@107d517c00 @ 548
/sys/class/gpio/gpiochip544 : gpio-brcmstb@107d508520 @ 544
``` 

I think my pins actually start at 571, however, I have not had time to figure out if I need a pull up resistor or something as the check with the volt meter showed nothing changing on 571 and 572.  The output of the cat would show that 571 and 572 are changing though.
```
 cat /sys/kernel/debug/gpio |grep gpio
gpiochip0: GPIOs 512-543, parent: platform/107d508500.gpio, gpio-brcmstb@107d508500:
 gpio-512 (-                   )
 gpio-513 (2712_BOOT_CS_N      |spi10 CS0           ) out hi ACTIVE LOW
 gpio-514 (2712_BOOT_MISO      )
 gpio-515 (2712_BOOT_MOSI      )
 gpio-516 (2712_BOOT_SCLK      )
 gpio-517 (-                   )
 gpio-518 (-                   )
 gpio-519 (-                   )
 gpio-520 (-                   )
 gpio-521 (-                   )
 gpio-522 (-                   )
 gpio-523 (-                   )
 gpio-524 (-                   )
 gpio-525 (-                   )
 gpio-526 (PCIE_SDA            )
 gpio-527 (PCIE_SCL            )
 gpio-528 (-                   )
 gpio-529 (-                   )
 gpio-530 (-                   )
 gpio-531 (-                   )
 gpio-532 (PWR_GPIO            |pwr_button          ) in  hi ACTIVE LOW
 gpio-533 (2712_G21_FS         )
 gpio-534 (-                   )
 gpio-535 (-                   )
 gpio-536 (BT_RTS              )
 gpio-537 (BT_CTS              )
 gpio-538 (BT_TXD              )
 gpio-539 (BT_RXD              )
 gpio-540 (WL_ON               |wl_on_reg           ) out hi
 gpio-541 (BT_ON               |shutdown            ) out hi
 gpio-542 (WIFI_SDIO_CLK       )
 gpio-543 (WIFI_SDIO_CMD       )
gpiochip1: GPIOs 544-547, parent: platform/107d508500.gpio, gpio-brcmstb@107d508520:
 gpio-544 (WIFI_SDIO_D0        )
 gpio-545 (WIFI_SDIO_D1        )
 gpio-546 (WIFI_SDIO_D2        )
 gpio-547 (WIFI_SDIO_D3        )
gpiochip2: GPIOs 548-564, parent: platform/107d517c00.gpio, gpio-brcmstb@107d517c00:
 gpio-548 (RP1_SDA             )
 gpio-549 (RP1_SCL             )
 gpio-550 (RP1_RUN             |RP1 RUN pin         ) out hi
 gpio-551 (SD_IOVDD_SEL        |vdd-sd-io           ) out hi
 gpio-552 (SD_PWR_ON           |sd_vcc_reg          ) out hi
 gpio-553 (SD_CDET_N           |cd                  ) in  lo ACTIVE LOW
 gpio-554 (SD_FLG_N            )
 gpio-555 (-                   )
 gpio-556 (2712_WAKE           )
 gpio-557 (2712_STAT_LED       |ACT                 ) out hi ACTIVE LOW
 gpio-558 (-                   )
 gpio-559 (-                   )
 gpio-560 (PMIC_INT            )
 gpio-561 (UART_TX_FS          )
 gpio-562 (UART_RX_FS          )
 gpio-563 (-                   )
 gpio-564 (-                   )
gpiochip3: GPIOs 565-570, parent: platform/107d517c00.gpio, gpio-brcmstb@107d517c20:
[... 160 lines cut ...]
202: op dh pd | hi // RP1_RUN/AON_GPIO2 = output
203: op dh pd | hi // SD_IOVDD_SEL/AON_GPIO3 = output
204: op dh pd | hi // SD_PWR_ON/AON_GPIO4 = output
205: a6    pu | lo // SD_CDET_N/AON_GPIO5 = SD_CARD_PRES_G
206: ip    pu | hi // SD_FLG_N/AON_GPIO6 = input
207: ip    pd | lo // AON_GPIO7 = input
208: ip    pd | lo // 2712_WAKE/AON_GPIO8 = input
209: op dh pd | hi // 2712_STAT_LED/AON_GPIO9 = output
210: ip    pd | lo // AON_GPIO10 = input
211: ip    pd | lo // AON_GPIO11 = input
212: ip    pd | lo // PMIC_INT/AON_GPIO12 = input
213: a3    pd | hi // UART_TX_FS/AON_GPIO13 = VC_TXD0
214: a3    pd | lo // UART_RX_FS/AON_GPIO14 = VC_RXD0
215: ip    pd | lo // AON_GPIO15 = input
216: ip    pu | hi // AON_GPIO16 = input
232: a1    -- | hi // HDMI0_SCL/AON_SGPIO0 = HDMI_TX0_BSC_SCL
233: a1    -- | hi // HDMI0_SDA/AON_SGPIO1 = HDMI_TX0_BSC_SDA
234: a1    -- | hi // HDMI1_SCL/AON_SGPIO2 = HDMI_TX1_BSC_SCL
235: a1    -- | hi // HDMI1_SDA/AON_SGPIO3 = HDMI_TX1_BSC_SDA
236: a2    -- | hi // PMIC_SCL/AON_SGPIO4 = BSC_M2_SCL
237: a2    -- | hi // PMIC_SDA/AON_SGPIO5 = BSC_M2_SDA

``` 

Hope this helps!!  Brian

istnv, 2024-07-22:
@Brian-TV 
That helps a lot. I can look for the `gpiochip` that is labeled `pinctrl-xxx` and read the `base` offset from there.
GPIO0 and GPIO1 are in the 'specs' but I do not see them actually on a header anywhere.
I'll have the module will add 'base' to the requested GPIO number. 
You will need to re-configure again, but this should end up matching the `pinctrl` values and not break the older kernels.

istnv, 2024-07-22:
@Brian-TV 
Can you test from the PR or do I need to publish in the betas?

Brian-TV, 2024-07-22:
How many files are you changing?  If it is just 1 or 2, I can probably handle that for a test.

istnv, 2024-07-22:
Three including the version # in `package.json`

Brian-TV, 2024-07-22:
Where is config.js located at?

istnv, 2024-07-22:
I created a v1.3.0 release you can download.

closed 2024-07-22 by istnv

Brian-TV, 2024-07-22:
Thanks!  I'm not 100% sure how to grab and install just that update.  I see the updated files on github.

Do I need to clone into it and pull it across that way?

istnv, 2024-07-22:
Have you used the local developer directory? 
You can download the zip and extract it into there.

istnv, 2024-07-22:
It will be in the next beta release. Those generate automatically about every 8 hours. May be sooner.
Then you can run `companion-update` on the pi and select the latest beta. 
EXPORT YOUR CONFIG FIRST. Then if there is a problem you can revert to an earlier release.

Brian-TV, 2024-07-22:
Thanks!!  I'll let you know when I get it updated!!

Brian-TV, 2024-07-25:
Hey @istnv 

I ran the update to the latest beta tonight and I'm getting this error in my error log...

```
Pin_num: 5 OUTPUT
error: Trace: Init failed: RangeError: Map maximum size exceeded
    at /opt/companion/bundled-modules/raspberry-gpio/main.js:1:443125
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async /opt/companion/bundled-modules/raspberry-gpio/main.js:1:412083
error: Failed to init: Error: Call timed out Error: Call timed out
    at t.IpcWrapper.sendWithCb (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:425869)
    at /opt/companion/bundled-modules/raspberry-gpio/main.js:1:442980
    at /opt/companion/bundled-modules/raspberry-gpio/main.js:1:411922
    at t.default._tryToStartAnother (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:410969)
    at /opt/companion/bundled-modules/raspberry-gpio/main.js:1:412129
    at new Promise (<anonymous>)
    at t.default.add (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:411772)
    at c._handleInit (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:442505)
    at t.IpcWrapper.receivedMessage (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:426338)
    at process.<anonymous> (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:441989)
system: ** Connection stopped **
system: ** Starting Connection from "/opt/companion/bundled-modules/raspberry-gpio/main.js" **
system: ** Connection started **
Starting up module class: c
Sentry disabled
Module-host accepted registration
debug: GPIO is accessible
info: Processing config
[ 0, 1, 2, 3, 4, 5 ]
Map(0) {}
[
  { id: 0, label: '0' },
  { id: 1, label: '1' },
  { id: 2, label: '2' },
  { id: 3, label: '3' },
  { id: 4, label: '4' },
  { id: 5, label: '5' }
]
Pin_num: 0 OUTPUT
info: Setting up GPIO
Pin_num: 1 OUTPUT
Pin_num: 2 OUTPUT
Pin_num: 3 OUTPUT
Pin_num: 4 OUTPUT
Pin_num: 5 OUTPUT
error: Trace: Init failed: RangeError: Map maximum size exceeded
    at /opt/companion/bundled-modules/raspberry-gpio/main.js:1:443125
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async /opt/companion/bundled-modules/raspberry-gpio/main.js:1:412083
error: Failed to init: Error: Call timed out Error: Call timed out
    at t.IpcWrapper.sendWithCb (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:425869)
    at /opt/companion/bundled-modules/raspberry-gpio/main.js:1:442980
    at /opt/companion/bundled-modules/raspberry-gpio/main.js:1:411922
    at t.default._tryToStartAnother (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:410969)
    at /opt/companion/bundled-modules/raspberry-gpio/main.js:1:412129
    at new Promise (<anonymous>)
    at t.default.add (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:411772)
    at c._handleInit (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:442505)
    at t.IpcWrapper.receivedMessage (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:426338)
    at process.<anonymous> (/opt/companion/bundled-modules/raspberry-gpio/main.js:1:441989)
system: ** Connection stopped **
```

istnv, 2024-07-25:
ugh. I really dislike trying to unpack the code.
I may know what is happening. It's going be a bit. My Pi4 is not very fast and I need to load a non-compact source.

istnv, 2024-07-25:
@Brian-TV 
Latest update should be in betas soon. Give it another try.

Brian-TV, 2024-07-25:
@istnv 

I'm not saying that I did it right, but I did find the dev folder.  Here are my results on Beta v3.4.0+7152-main-7e8a73c2:

```
debug: Only recent lines are shown here, nothing is persisted
system: ** Starting Connection from "/opt/companion-module-dev/companion-module-raspberry-gpio/lib/main.js" **
system: ** Connection started **
error: node:internal/errors:496
    ErrorCaptureStackTrace(err);
    ^

Error [ERR_MODULE_NOT_FOUND]: Cannot find package '@companion-module/base' imported from /opt/companion-module-dev/companion-module-raspberry-gpio/lib/main.js
    at new NodeError (node:internal/errors:405:5)
    at packageResolve (node:internal/modules/esm/resolve:916:9)
    at moduleResolve (node:internal/modules/esm/resolve:973:20)
    at defaultResolve (node:internal/modules/esm/resolve:1193:11)
    at ModuleLoader.defaultResolve (node:internal/modules/esm/loader:404:12)
    at ModuleLoader.resolve (node:internal/modules/esm/loader:373:25)
    at ModuleLoader.getModuleJob (node:internal/modules/esm/loader:250:38)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/module_job:76:39)
    at link (node:internal/modules/esm/module_job:75:36) {
  code: 'ERR_MODULE_NOT_FOUND'
}

Node.js v18.20.2
system: ** Connection stopped **
```

carmino2k, 2024-07-25:
Thank you all for considering my report.
My structure is different and I am reposting it. I hope it can help to make the update more compatible with all versions of raspberry. Thank you !!!

carmino@CompanionPi:~ $ cat /sys/kernel/debug/gpio |grep gpio
 gpiochip0: GPIOs 512-565, parent: platform/3f200000.gpio, pinctrl-bcm2835:
 gpio-512 (ID_SDA              )
 gpio-513 (ID_SCL              )
 gpio-514 (GPIO2               )
 gpio-515 (GPIO3               |shutdown            ) in  hi IRQ ACTIVE LOW
 gpio-516 (GPIO4               )
 gpio-517 (GPIO5               )
 gpio-518 (GPIO6               )
 gpio-519 (GPIO7               )
 gpio-520 (GPIO8               )
 gpio-521 (GPIO9               )
 gpio-522 (GPIO10              )
 gpio-523 (GPIO11              )
 gpio-524 (GPIO12              )
 gpio-525 (GPIO13              )
 gpio-526 (GPIO14              )
 gpio-527 (GPIO15              )
 gpio-528 (GPIO16              )
 gpio-529 (GPIO17              )
 gpio-530 (GPIO18              )
 gpio-531 (GPIO19              )
 gpio-532 (GPIO20              )
 gpio-533 (GPIO21              )
 gpio-534 (GPIO22              )
 gpio-535 (GPIO23              )
 gpio-536 (GPIO24              )
 gpio-537 (GPIO25              )
 gpio-538 (GPIO26              )
 gpio-539 (GPIO27              )
 gpio-540 (NC                  )
 gpio-541 (LAN_RUN_BOOT        )
 gpio-542 (CTS0                )
 gpio-543 (RTS0                )
 gpio-544 (TXD0                )
 gpio-545 (RXD0                )
 gpio-546 (SD1_CLK             )
 gpio-547 (SD1_CMD             )
 gpio-548 (SD1_DATA0           )
 gpio-549 (SD1_DATA1           )
 gpio-550 (SD1_DATA2           )
 gpio-551 (SD1_DATA3           )
 gpio-552 (PWM0_OUT            )
 gpio-553 (PWM1_OUT            )
 gpio-554 (ETH_CLK             )
 gpio-555 (WIFI_CLK            )
 gpio-556 (SDA0                )
 gpio-557 (SCL0                )
 gpio-558 (SMPS_SCL            )
 gpio-559 (SMPS_SDA            )
 gpio-560 (SD_CLK_R            )
 gpio-561 (SD_CMD_R            )
 gpio-562 (SD_DATA0_R          )
 gpio-563 (SD_DATA1_R          )
 gpio-564 (SD_DATA2_R          )
 gpio-565 (SD_DATA3_R          )
gpiochip1: GPIOs 566-567, parent: platform/soc:firmware:virtgpio, brcmvirt-gpio, can sleep:
 gpio-566 (                    |ACT                 ) out lo
gpiochip2: GPIOs 568-575, parent: platform/soc:firmware:expgpio, raspberrypi-exp-gpio, can sleep:
 gpio-568 (BT_ON               )
 gpio-569 (WL_ON               )
 gpio-570 (STATUS_LED          )
 gpio-571 (LAN_RUN             )
 gpio-572 (HDMI_HPD_N          |hpd                 ) in  lo ACTIVE LOW
 gpio-573 (CAM_GPIO0           |cam1_regulator      ) out lo
 gpio-574 (CAM_GPIO1           )
 gpio-575 (PWR_LOW_N           |PWR                 ) in  hi
carmino@CompanionPi:~ $

Brian-TV, 2024-07-25:
@istnv 

Ignore the previous message.

Ok, I got another SD card so I could really mess around with it.  I installed beta v3.4.0+7122-main-3bc10380 ...

Now I learned how to install node and yarn and all.  I ran yarn in that directory and it initially looked like it was going to work.  However, after trying to assign Gpio 0, I could not see the pin number in the buttons.  The drop down list just reads "undefined"  As long as I don't shut it off and restart the Raspberry Pi GPIO module, it looks as it is running.  Once I stop it and restart, it starts crashing.

I think I've got the understanding on how to install in the dev.  So, I should be able to test things a bit more quickly.

Here is the log:
```
info: Initialising actions
info: Initialising feedbacks
info: Initialising variables
info: Initialising presets
Generating config fields...
debug: GPIO is accessible (base at 571)
info: Processing config
[ 0 ]
Map(0) {}
[ { id: 0, label: '0' } ]
Pin_num: 0 OUTPUT
info: Setting up GPIO
system: ** Connection stopped **
system: ** Starting Connection from "/opt/companion-module-dev/companion-module-raspberry-gpio/lib/main.js" **
system: ** Connection started **
Starting up module class: GpioInstance
Sentry disabled
gpiochip571 571
Module-host accepted registration
debug: GPIO is accessible (base at 571)
info: Processing config
[ 0 ]
Map(0) {}
[ { id: 0, label: '0' } ]
info: Setting up GPIO
Pin_num: 0 OUTPUT
error: Failed to init: Error: Call timed out Error: Call timed out
    at t.IpcWrapper.sendWithCb (/opt/companion/main.js:2:3053831)
    at x.init (/opt/companion/main.js:2:3486022)
    at l.o (/opt/companion/main.js:2:3497520)
    at l.emit (node:events:517:28)
    at ChildProcess.<anonymous> (/opt/companion/main.js:2:3040069)
    at ChildProcess.emit (node:events:517:28)
    at emit (node:internal/child_process:944:14)
    at process.processTicksAndRejections (node:internal/process/task_queues:83:21)
system: ** Connection stopped **
``` 

Thanks!  Brian

Brian-TV, 2024-07-25:
@istnv 

First of all, THANKS! for working on this!!

Ok, that fixed the crash issue.  

When I click to enable a pin in the Gpio configuration, I see that pin on the Pi 5 change states as they usually do.

I can now assign a button to a Gpio.

However, when commanding the button to change the Gpio state, nothing is happening on the Pi 5. I tried IO0 (571), IO2 (573), and IO3 (574).  

Something must not be connected across?  Also, the feedback is not showing either state as I have the background set to black. Feedback is set to High = green and Low = red.  It is showing black.

Here is my log:

```
debug: Only recent lines are shown here, nothing is persisted
system: ** Connection stopped **
system: ** Starting Connection from "/opt/companion-module-dev/companion-module-raspberry-gpio/lib/main.js" **
system: ** Connection started **
Starting up module class: GpioInstance
Sentry disabled
gpiochip571 571
Module-host accepted registration
debug: GPIO is accessible (base at 571)
info: Processing config
[ 0, 2, 3 ]
Map(0) {}
[ { id: 0, label: '0' }, { id: 2, label: '2' }, { id: 3, label: '3' } ]
info: Setting up GPIO
Pin_num: 0(571) OUTPUT
Pin_num: 2(573) OUTPUT
Pin_num: 3(574) OUTPUT
info: GPIO Setup complete
info: Started polling regularly
info: Initialising actions
[]
info: Initialising feedbacks
info: Initialising variables
info: Initialising presets
Generating config fields...
info: Setting pin 3 to LOW (0)
info: Setting pin 3 to HIGH (1)
system: ** Connection stopped **
```

istnv, 2024-07-26:
Getting closer. Pin 0 and 1 are not exposed on the header, but I would expect GPIO3 on Pin 3 to work.
Hmmm. I'll dive in again. I may need to take my production Pi off line to test better.

Brian-TV, 2024-07-26:
@istnv 

No more errors, but it isn't controlling the pins or giving pin feedback.

It is however changing the state of the configured pins after boot.  So they start high and when I start the module, the configured ones go low.  So we're gaining!!

```
system: ** Starting Connection from "/opt/companion-module-dev/companion-module-raspberry-gpio/lib/main.js" **
system: ** Connection started **
Starting up module class: GpioInstance
Sentry disabled
gpiochip571 571
Module-host accepted registration
debug: GPIO is accessible (base at 571)
info: Processing config
[ 0, 2, 3 ]
Map(1) {
  27 => {
    trigger_type: 'none',
    debounce_enable: false,
    debounce_time: 10,
    invert_values: false
  }
}
[
  { id: 0, label: '0' },
  { id: 2, label: '2' },
  { id: 3, label: '3' },
  { id: 27, label: '27' }
]
info: Setting up GPIO
Pin_num: 0(571) OUTPUT
Pin_num: 2(573) OUTPUT
Pin_num: 3(574) OUTPUT
Pin_num: 27(598) INPUT
Trigger_type: none
Debounce_enable: false
Debounce_time: 10
Invert_values: false
info: GPIO Setup complete
info: Started polling regularly
info: Initialising actions
[]
info: Initialising feedbacks
info: Initialising variables
info: Initialising presets
Generating config fields...
info: Setting pin 3 to HIGH (1)
```

Brian-TV, 2024-07-27:
@istnv 

Yup, Yup, Yup.  That is working exactly like it was when I changed the numbers in the file up. 

Once I cloned into the github and put it in the dev directory on the stable version that I already had set up.. obviously, I had to change every pin number, but it does work as expected!!!

Thank you!! Thank you!! again!!!!

FYI about the Pi 5 ..  IO0 and IO1 do work (571 and 572)  They are oddly labeled differently on every breakout that I have.  They are a bit in lower than the middle, saying that the USB and Ethernet ports are the bottom. Pins 27 and 28.

I'm still learning how things are passed along in the files.  Eventually I need to take the grassvalley-amp module and modify it up a little bit to use with my (Leitch > Harris > Imagine) Nexio video servers.  I need hex output, not string (plus there are other differences on how they initialize the machine).  I'm still trying to figure out which modules do what at this point.  For now, I'm just using the tcp-udp sender to send hex out on tcp port 557.  So, it is working, but I'm not getting any feedback, but I wasn't really getting feedback before with GPIO connections! (it either played or it didn't!)

Thanks again!!  Brian

istnv, 2024-07-27:
@carmino2k @dani1810 
The module now locates the offset number for the GPIO pin range and automatically adjusts the internal numbers.
This fix is in the beta release.

PR #21 (2024-07-22): Acquire pin offset base for newer OS kernels
Newer kernels shift the gpio pins to a number greater than zero.
This adds a pin number offset, extracted from the O/S configuration.
This offset is applied to each API pin request.
Should fix #20
PR #24 (2024-07-25): Fix a 'double' offset calculation that crashed the module
Another attempt to fix #20